sameer/svg2gcode #3: Add Basic Documentation & Tests

lorenrogers, 2020-03-21:
This project is great, and I'd love to contribute to it! The main issue I have is that nothing is documented or tested.. The README is pretty good, but the code itself doesn't have inline docs, and there's no unit testing. 

To get this off the ground, I suggest adding a simple baseline level of documentation for each function stating what it's intended to do, and adding unit tests for the most critical functions.

sameer, 2020-03-21:
I completely agree, I originally wrote this just for using on a pen plotter I made for a class project. I didn't realize how much time it would take so I added a readme for others to use it.

I'll take some time to try and document what I still remember next week.

If you have any ideas for a testing harness let me know. I thought maybe there could be a few mock device profiles that the output gcode is run on and see if the head crashes or any errors occur.

Any contributions you might have are welcome!
-------- Original Message --------

lorenrogers, 2020-03-22:
There's a good amount of work in here, and it seems to be the most up to date SVG to Gcode interpreter I could find. (Everything else I've looked at is either abandoned or unreadable spaghetti code.) I think a basic set of unit tests would be enough to get started here. I'll draft some quick documentation outlines to get the process started, and put a simple unit test framework to close out this ticket. I mostly wanted to open the conversation here.