sameer/svg2gcode #7: SVG offset

jeevank, 2020-05-23:
Is there a way to get all of the gcode offset by a certain amount on both X and Y axis?

Without the offset the gcode starts the printer at the very edge of the print bed. Thanks!

sameer, 2020-05-23:
Hi @jeevank, good idea. I hadn't thought about this before, I can add CLI arguments for manipulating the SVG placement. How does this sound?

```
--origin <x,y> place the upper left corner of the SVG
```
That way you can always place it within bounds.

jeevank, 2020-05-24:
There are 2 free Inkscape plugins that allow generating Gcode (J Tech Phototonics laser) and Inkscape's built-in "Gcodetools". Both use the bottom-left corner as the 0,0 X,Y offset so it may be good to align with that.

But I'm personally happy with either choice. Thanks for the quick response.

(To provide some context, I'm currently using a method similar to the one here to use a 3d printer as a plotter:
https://medium.com/@urish/how-to-turn-your-3d-printer-into-a-plotter-in-one-hour-d6fe14559f1a)

sameer, 2020-05-24:
Thanks for sharing this, that sounds like a good convention to follow.
And that looks pretty cool!

-------- Original Message --------

sameer, 2020-05-24:
Hello Jeevan,

Please try the latest commit when you have some time. Here's the demo from the README with the origin argument:
```
cargo run --release -- examples/Vanderbilt_Commodores_logo.svg --origin 45,46 --off 'M4' --on 'M5' -o out.gcode
```

jeevank, 2020-05-28:
This worked perfectly! Thanks for adding this @sameer 🥇

sameer, 2020-05-28:
Great! In that case, I will close this issue.

closed 2020-05-28 by sameer