sameer/svg2gcode #9: off commands not part of output gcode

sloev, 2021-03-03:
i am running the following command with one of the examples:
```
$ cargo run --release -- examples/Vanderbilt_Commodores_logo.svg --off 'M5' --on 'M3 S1000' -o out.gcode
```

it generates the following gcode:

```
G21
G94
G90
M3 S1000
G0 X0 Y0.3942308528309013;svg > g > g > path
G0 X58.644520553969095 Y113.10977699532148
M3 S1000
G1 X43.78657986962459 Y72.10494540682015 F300
G1 X8.218721863111911 Y72.11494512188344 F300
G1 X0.669752000000008 Y72.1169450648961 F300
G1 X34.73061602271223 Y46.238682452624076 F300
G1 X17.206685981491805 Y0 F300
G1 X58.48552118872407 Y24.83729227418418 F300
G1 X92.9503835985998 Y4.149881751262754 F300
G1 X99.86235600472372 Y0 F300
G1 X82.33942595951112 Y46.238682452624076 F300
G1 X116.3992899862155 Y72.1169450648961 F300
G1 X108.85052012230517 Y72.11494512188344 F300
G1 X73.28266211579249 Y72.10514540112143 F300
G1 X61.212710301279216 Y105.9741803204635 F300
G1 X58.64472055317068 Y113.11027698107469 F300
G1 X58.644520553969095 Y113.10977699532148 F300;svg > g > g > path
M3 S1000
G0 X58.64052056993776 Y105.1803229410195
M3 S1000
G1 X45.70857219667466 Y69.49533976618511 F300
G1 X8.218721863111897 Y69.5053394812484 F300
G1 X37.833603635010036 Y47.00398064584731 F300
G1 X22.597664459705626 Y6.799126262408663 F300
G1 X58.48952117275537 Y28.39351094156993 F300
G1 X94.470377530502 Y6.799126262408677 F300
G1 X79.23343835918978 Y47.003980645847335 F300
G1 X108.84932012709574 Y69.50533948124843 F300
G1 X71.35846979752513 Y69.49533976618514 F300
G1 X58.64052056993773 Y105.18032294101954 F300
G1 X58.64052056993776 Y105.1803229410195 F300;svg > g > g > path
M3 S1000
G0 X43.35758158226533 Y67.00581070402998
M3 S1000
G1 X43.35758158226533 Y61.70906163215922 F300
G1 X46.04307086129395 Y61.70906163215922 F300
G1 X54.470737216486455 Y30.691945448865695 F300
G1 X61.83010783651371 Y30.69194544886571 F300
G1 X70.859371790017 Y61.709061632159255 F300
G1 X73.4511614431119 Y61.709061632159255 F300
G1 X73.4511614431119 Y67.00581070403003 F300
G1 X62.293205987739974 Y67.00581070403004 F300
G1 X62.293205987739974 Y61.709061632159276 F300
G1 X65.21119433858917 Y61.709061632159276 F300
G1 X58.58822077872823 Y37.53175055215182 F300
G1 X52.47884516848901 Y61.70906163215927 F300
G1 X55.44173334008978 Y61.70906163215927 F300
G1 X55.44173334008978 Y67.00581070403004 F300
G1 X43.35778158146689 Y67.00581070403004 F300
G1 X43.35758158226533 Y67.00581070402998 F300
M3 S1000
G0 X0 Y0.3942308528309013
M20
```

as you can see it does not include the "off" commands in its output, any idea why? @sameer 
btw: this little tool has singlehandedly gotten me to my first output so thanx a bunch!

sameer, 2021-03-03:
Hey sloev,

I recently refactored the code I wrote for gcode parsing and I set the tool off action to be the tool on action. :man_facepalming: 

I'll be looking into adding tests that cover these and other edge cases so it doesn't happen again

closed 2021-03-03 by sameer

sloev, 2021-03-04:
@sameer hey thanks for replying!!!
btw i fixed my situation by using a 3 line postprocessor that alternates between off/on, but having it __in the tool__ is preferable!
bt the way here is something i was able to make thanks to your software!!! (thanks a bunch!)
[image]

sloev, 2021-03-04:
btw i can help with creating a basic test suite if you want it @sameer ?
it will take me no time

sameer, 2021-03-04:
Sure, that would be helpful!

sameer, 2021-03-05:
I missed seeing that picture you shared when I replied by email :hushed: 
what does the SVG look like?

commit a72ab629 (2021-03-03): use tool off action on tool off, fixes #9

Will be adding tests for this behavior sometime this week