sameer/svg2gcode #15: Fill shape

FPGA-KING, 2021-06-28:
Hello I want to fill the shape.

Is it possible to draw filled shape?

For example If I draw star image as like example 

I want the library to make automatically tool path for fill the star.

sameer, 2021-06-28:
Hi there,

svg2gcode does not support fills at the moment ([see FAQ](https://github.com/sameer/svg2gcode#faq--interesting-details)).

What kind of fill are you looking to do? Definitely open to collaborating on this :slightly_smiling_face: 

Thanks,
Sameer

FPGA-KING, 2021-07-03:
Raster path

sameer, 2021-07-04:
That seems like a reasonable use case. Is this something you'd be interested in making a pull request for?

FPGA-KING, 2021-07-04:
> That seems like a reasonable use case. Is this something you'd be interested in making a pull request for?

Yes, It is. I have idea for it. but the problem is that I can use CPP only. 

I will first convert it to cpp code

sameer, 2021-07-04:
Cool, this is very exciting. Happy to assist in the translation to Rust process.

sameer, 2023-05-30:
>  the plotter raises the Z-axis and does not leave connection lines?

You can add in some g-code for that as the tool off sequence. 
[image]

Is this what you were looking for?

chrisgwgreen, 2025-06-16:
Any developments on the fill feature?