sameer/svg2gcode #18: Strange plot points on curves from SVG conversion

Milner99, 2021-09-02:
[happy dog svg and gcode.zip](https://github.com/sameer/svg2gcode/files/7100766/happy.dog.svg.and.gcode.zip)

Hi,

Attached is an SVG and GCode output from the online demo of svg2gcode. Seeing some strange plotted points near the head of the cartoon dog. Is this something going on in the SVG or the conversion?

Would love some advise and guidance please. thank you so much
Chris.

sameer, 2021-09-02:
Hi Chris,

Thank you for bringing this to my attention. Just pushed changes that fix the strange plot points:

[image]

This was a miss on my part when refactoring svg2gcode to support circular interpolation.

* Root cause: incorrect multiplication in control point tracking
* Action taken: 
    * Correct the issue and push changes for that
    * Add test cases using the provided image to ensure this particular use case does not see future regressions
    * Making plans to test other edge cases based on coverage reports

closed 2021-09-02 by sameer

sameer, 2021-09-02:
I have confirmed that those changes have reached the web interface.

Thank you for using svg2gcode! Please let me know if you run into any other problems :slightly_smiling_face:

Milner99, 2021-09-02:
Wow, quick work. Thank you so much!

commit 69287b28 (2021-09-02): fix #18: incorrect multiplication on smooth curve control points