sameer/svg2gcode #20: Converting from 'Percent' to millimeters is not supported, treating as millimeters

matopeto, 2021-10-02:
Hi

I exported svg from affinity and svg looks like:

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg width="100%" height="100%" viewBox="0 0 354 1181" version="1.1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" xml:space="preserve" xmlns:serif="http://www.serif.com/" style="fill-rule:evenodd;clip-rule:evenodd;stroke-linecap:round;stroke-linejoin:round;stroke-miterlimit:1.5;">
    <g transform="matrix(0.30102,0,0,1,4.19388,0)">
        <path d="M6,1175L1182,1179L1178,3L6,1L6,1175Z" style="fill:none;stroke:rgb(255,0,57);stroke-width:1.13px;"/>
    </g>
</svg>
```

It defines `viewBox`  (in some units, seems like pixels mm x dpi) and `width` and `height` in percent

I run your program and get this warning

`Converting from 'Percent' to millimeters is not supported, treating as millimeters`

And result is 100x100 mm instead correct dimensions

It can be sized by `--dimensions` but i cannot specify only one dimension because of wrong aspect ratio 100x100 instead of 354x1181

sameer, 2021-10-03:
Hi matopeto,

I've added some more nuanced handling for the use case you described:

* Percentage conversion is now supported. Preserves aspect ratio and treats the viewBox as being in millimeters (i.e. 100% x 100% = 354mm x 1181mm)
* Specifying a viewBox and no width/height assumes the viewBox is in millimeters

closed 2021-10-03 by sameer

matopeto, 2021-10-10:
It works, great :)

commit f55ab958 (2021-10-03): add %/em conversion & assume viewBox in mm (fixes #20)