sameer/svg2gcode #21: DPI is not working (or maybe wrong parsing input dimensions)

matopeto, 2021-10-02:
I exported SVG from affinity (with without viewBox option)

SVG look like this:

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg width="354px" height="1181px" version="1.1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" xml:space="preserve" xmlns:serif="http://www.serif.com/" style="fill-rule:evenodd;clip-rule:evenodd;stroke-linecap:round;stroke-linejoin:round;stroke-miterlimit:1.5;">
    <g transform="matrix(0.30102,0,0,1,4.19388,0)">
        <path d="M6,1175L1182,1179L1178,3L6,1L6,1175Z" style="fill:none;stroke:rgb(255,0,57);stroke-width:1.13px;"/>
    </g>
</svg>
```

so `width="354px" height="1181px" ` it is cca 3x10cm with 300 DPI 

i run svg2gcode and specify `--dpi=300`

```
cargo run --release -- examples/affinityexport-wo-viewbox.svg --dpi=300 --off 'M4' --on 'M5' -o out.gcode
```

And it generates:

```
G21
G90;svg > g > path
M4
G0 X17263.872 Y38396.67199999839
M5
G1 X1035830.93889024 Y0 F300
G1 X1032366.42505728 Y11288621.567999998 F300
G1 X17263.872 Y11307819.904 F300
G1 X17263.872 Y38396.67199999839 F300
M4
M2
```

Values are very high `1035830.93889024`

Also if i run command without DPI option (it should use 96 default)

```
cargo run --release -- examples/affinityexport-wo-viewbox.svg --off 'M4' --on 'M5' -o out.gcode
```

But it generates also "nonsense" values:

```
G21
G90;svg > g > path
M4
G0 X53949.6 Y119989.59999999404
M5
G1 X3236971.684032 Y0 F300
G1 X3226145.078304 Y35276942.39999999 F300
G1 X53949.6 Y35336937.199999996 F300
G1 X53949.6 Y119989.59999999404 F300
M4
M2
```

sameer, 2021-10-03:
Hi matopeto,

The lack of a viewBox attribute is definitely causing some issues here. Do you know why it is missing? Looking at the [W3 specs](https://www.w3.org/TR/SVG11/coords.html#ViewBoxAttribute) I see:

> All elements that establish a new viewport (see elements that establish viewports), plus the ‘marker’, ‘pattern’ and ‘view’ elements have attribute ‘viewBox’. 
>
> ...
> The following elements establish new viewports:
> * The ‘svg’ element
> ...

So svg2gcode expects a viewBox to be present and uses that for some calculations.

matopeto, 2021-10-04:
@sameer OK, thanks for explenation. So if svg without viewbox is not supported, maybe showing some error would be appropriate.

>  Do you know why it is missing?
Viewbox in my svg is missing becase i check "without viewbox" checkbox in export. I was thinking it will be workaround for #20

sameer, 2021-10-05:
Good point, I'll add some logic to make that required.

sameer, 2024-03-08:
This should be resolved after https://github.com/sameer/svg2gcode/commit/8f9db323c5753f3ee88b6041becb1ee25ac7ba00 which now accounts for viewboxes that don't exist.

I plan to publish and release this soon

closed 2024-03-08 by sameer