sameer/svg2gcode #28: Straight lines not being converted to gcode

ln0a, 2022-10-08:
First off I just want to say your tool is brilliant. Thanks so much for developing it!
I'm having one small problem though, that if the svg path does not contain any curves, no gcode will be created.
I have tried this as both open and closed paths and get the same result, and on the web and cli releases.

Any thoughts on why this might be?

sameer, 2022-10-10:
Hi, glad to hear it's been helpful so far!

Do you mind sharing the SVG you used or a similar example so I can look into this further?

ln0a, 2022-10-10:
Of course, thanks for having a look at this! I have found a solution, 'making compound paths' in the illustrator allows the gcode to be generated, but when trying to automate 100s of svgs this feels only a temporary solution.
[image]

sameer, 2022-10-11:
I think I see the problem, this is related to #15. The SVG you shared contains a polygon object, whereas svg2gcode expects it to be a path:

```
<polygon class="st0" points="429.7,425.2 221,526.3 429.7,561 "/>
```
becomes
```
<path
   id="polygon4"
   class="st0"
   style="fill:none;stroke:#000000;stroke-miterlimit:10"
   d="M 429.7,425.2 221,526.3 429.7,561 Z" />
```
using Inkscape's Object to Path.

It should be possible to add support for these, they just aren't handled currently. Though they will not be filled until #15 is implemented.

sameer, 2023-06-01:
Closing in favor of the mega tracker #38