sameer/svg2gcode #29: Problems with running CLI / batch SVG file processing

andszwabe, 2022-10-25:
Hey Sameer! 
Your tool seems to be perfect for what I need in the project, but I need to use it on a larger batch of SVG files (3000), and to do so I need to figure out the CLI use. Unfortunately I can't get it to work - the details are in my question on StackOverflow (https://stackoverflow.com/questions/74193515/svg2gcode-generator-for-laser-cutting-in-rust-how-to-use-it-for-batch-svg-p )
I would be very much thankful for Your answear!
Cheers, 
Andrew

sameer, 2022-10-26:
Hi andszwabe,

If it's easier for you, the web UI can do bulk processing.

For the CLI, the usage notes should explain how to use it -- `cargo run --release -- --help`

andszwabe, 2022-10-26:
@sameer thank You for the answer! 
1. Indeed,(the web tool) now it works for batches! For some reason bulk processing didn't work when I tried it couple of times before, it picked just one file for exporting (I was on a Mac then but I don't think it matters - right?) 
2. For now I will assume that it will be enough for me and probably for the others to just use the web tool or to make their own solution for bulk processing in CLI, I have no real programming skills so this would be very difficult task for me now, but for others probably not. 
If I had any further problems I will write You an email not to clutter this post, thank You for Your offer!
Andrew

closed 2022-10-26 by andszwabe

sameer, 2022-10-26:
>  it picked just one file for exporting (I was on a Mac then but I don't think it matters - right?)

There might be some browser limitation...I can look into making the bulk processing download a ZIP file. How does that sound?

andszwabe, 2022-10-27:
> >  it picked just one file for exporting (I was on a Mac then but I don't think it matters - right?)
> 
> 
> 
> There might be some browser limitation...I can look into making the bulk processing download a ZIP file. How does that sound?

@sameer  That would be awesome! I tried it yesterday on Chrome / Windows and it processed about 80 files out of the 3000 I uploaded, and they were picked out of order too, so maybe that will tell You something about the problem... anyway - thank You for making this, Your tool will probably make my project a reality :)