sameer/svg2gcode #36: Origin based on svg 0,0

MadTooler, 2023-03-04:
I have only been using your website demo.  From there, when I load an SVG made in inkscape, it seems to default X 0 as the farthest left extent of geometry and Y 0 as the farthest bottom extent of geometry with no consideration of file 0,0 reference.  In general, that may be a good thing, and what I actually would prefer.  The exception is when you need to maintain the 0,0 as designed for alignment or other purposes.  

Is there currently a method to maintain file origin as the output origin (not looking to have to custom set the offsets every time)? 

Thanks

MadTooler, 2023-04-23:
Is there any chance of this feature?

If not, can you direct me to where in the code I may customize mine?

Thanks again.

crxksa, 2023-05-20:
any update?

please help

MadTooler, 2023-05-21:
@crxksa @sameer 

What I am finding, while generating svg files from Inkscape, is if I have set a front page size via the menu file>Document Properties>Display, and the geometry is either within the page or above the bottom and right of the left of the page, the gcode is output referencing the bottom left of the page.  If the geometry is left of the left edge of the page, the furthest left of the geometry is the gcode X0.  If the geometry is below the bottom edge of the page, the most bottom point of the geometry is the gcode Y0.  However, the geometry can go past the page right edge and top edge while still using the bottom left edge of the page as the gcode X0Y0 as long as there is no geometry beyond the left and bottom edges of page.  Does that make sense???

For me, it is pretty messy, but I can usually make do.  If I ever have a super huge geometry that is bigger than the max page size, things may get interesting, but I haven't hit that yet.

Maybe there should be a config option to output with origin to SVG 0,0 regardless of where it falls on the page???

Another either solution and/or feature would be the ability to have an origin point object within the svg so the pages coordinates are not driving and that origin object was the 0,0 reference.

sameer, 2023-05-26:
Sorry for the delay here both, things have been pretty busy. Will take a look at this today

sameer, 2023-05-26:
I had started an upgrade of the web version but it needs a bit of refactoring to work with yew. Once I get through that I'll start working on this

sameer, 2023-05-26:
This is actually the same problem as #7. Unlike the cli, the web interface [assumes you always want a valid floating point origin](https://github.com/sameer/svg2gcode/blob/master/web/src/state.rs#L13).

I will resolve this right after the yew update.

commit 99cde4ae (2023-06-01): update dependencies, fix #36
closed 2023-06-01 by sameer