sameer/svg2gcode #37: Install for CLI on RPI

MadTooler, 2023-04-22:
I would like to run the CLI on a raspberry pi 4b running Pi OS 64bit.  Is there any documentation of the installation of svg2gcode for this purpose?

For running CLI, is there documentation of the parameters?

Thanks!

MadTooler, 2023-04-23:
I haven't found any documentation on the install, but here is what I was able to make work.

Install rust and cargo with this script per https://doc.rust-lang.org/cargo/getting-started/installation.html
    `curl https://sh.rustup.rs -sSf | sh`
	
Clone the svg2gcode github repo
   `sudo git clone https://github.com/sameer/svg2gcode.git`

I had to set permissions of directories or else I was getting a Permission denied (os error 13):	
    `sudo chown -R $(whoami) /home/pi/svg2gcode/`
		
and again here, but I am not sure this one was needed:
    `sudo chown -R $(whoami) /home/pi/.cargo/  `

Note, on the first run of the cargo svg2gcode, it will take a long time as it builds the project.  Afterwards, it is quick.
	
Example CLI call if running from the root directory with example of an svg file in the desktop with an output to the desktop:
	
    `cargo  run --manifest-path svg2gcode/Cargo.toml  --release -- /home/pi/Desktop/drawing.svg --off 'M5' --on 'M4' -o /home/pi/out.gcode`

sameer, 2023-04-23:
Hi MadTooler,

This installation process is pretty standard if you are familiar with Rust. It is a little confusing if you aren't so I created the web version. I'll add some info to the readme.

There are some pre-built binaries but not for aarch64 yet. https://github.com/sameer/svg2gcode/releases/tag/v0.0.6

> I haven't found any documentation on the install, but here is what I was able to make work.
>
>
> Install rust and cargo with this script per [...]
`

MadTooler, 2024-08-06:
I am still not anymore familiar with rust as I only run your project from within it.  When updating to your latest git, what all needs to be run?

Thanks.

sameer, 2024-08-08:
> I am still not anymore familiar with rust as I only run your project from within it. When updating to your latest git, what all needs to be run?
> 
> Thanks.

Hey, sorry I missed this! Running the below should be sufficient to get the CLI built and running locally

```
cargo run --release -p svg2gcode-cli -- <ARGS>
```