sameer/svg2gcode #38: Support other non-closed, non-path elements

sameer, 2023-06-01:
#32 pointed out that polylines don't work. Line should also work https://www.w3.org/TR/SVG11/shapes.html#LineElement

kvdveer, 2024-01-02:
I've started work on this, I've written the code for `rect`, `circle`, `ellipse`, `line`, and `polyline`. I believe those are all the shapes from the spec.

Unfortunately I'm running into some problems with the coordinate system, I'm afraid I'll have to rework some of that according to [SVG1.1 § 7](https://www.w3.org/TR/SVG11/coords.html). Do you prefer the coordinate rework as a separate MR, or as part of this one?

sameer, 2024-01-02:
1 is fine, I appreciate you looking into this!

sameer, 2024-04-06:
Fixed by 225574d34a7a180a68cac443711671845966149a

closed 2024-04-06 by sameer