sameer/svg2gcode #42: question: does svg2gcode  support creating G02/G03 commands?

monperrus, 2023-11-17:
quick question: does svg2gcode support creating G02/G03 commands for efficient curves?

thanks!

sameer, 2023-11-17:
Hi, yes it does. It is still marked as experimental though since there's some room for improvement.

monperrus, 2023-11-18:
ack thanks!

closed 2023-11-18 by monperrus

monperrus, 2023-11-20:
I'm testing my SVG with curves but only get G01.
what's the flag to get G02/G03 output?

thanks!

sameer, 2023-11-20:
Hey, it should be `--circular-interpolation true`

monperrus, 2023-11-21:
Thanks a lot, documented at https://github.com/sameer/svg2gcode/pull/44

I confirm that we get G02 commands with this [benchmark file](https://github.com/bbaudry/swart-studio/blob/main/penplotting/curves_benchmark.svg). FTR, there are still lots of G01 remaining (2k+).