sameer/svg2gcode #43: cannot build unknown feature `proc_macro_span_shrink`

monperrus, 2023-11-18:
FYI cannot build today
```
$ cargo --version
cargo 1.73.0-nightly (694a57956 2023-07-11)

$ cargo build
   Compiling proc-macro-error v1.0.4
error[E0635]: unknown feature `proc_macro_span_shrink`
  --> /home/martin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.59/src/lib.rs:92:30
   |
92 |     feature(proc_macro_span, proc_macro_span_shrink)
   |                              ^^^^^^^^^^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0635`.
error: could not compile `proc-macro2` (lib) due to previous error
warning: build failed, waiting for other jobs to finish...
```

sameer, 2023-11-18:
Hey, I haven't tried compiling in a while but this should be fixed by updating the dependencies. I'll try to do that sometime this weekend, or if you'd like to make a pull request I would be happy to review it.

sameer, 2023-11-19:
Fixed by 915b747ec9de8acb17d7cba43d3866da521d3ead

closed 2023-11-19 by sameer

monperrus, 2023-11-20:
I confirm I can build now. Thanks a lot @sameer