sameer/svg2gcode #49: your code can produce single-line word?

ishangxue, 2024-01-31:
Hello, thank you for sharing svg2gcode, now there is a problem I hope to get your help,
[image]

 in SVG generated gcode, using a writer to write out the outline of the word,

[
[image]
](url)
 obviously this is not the reality of life, what we need in life is a single-line font, I try to use other tools to deal with the single-line, but not ideal, your code can produce single-line word?
[image]

sameer, 2024-01-31:
Hi @ishangxue, could you share the SVG file you are using? That way I can look into why the path is not smooth.

I wonder if this is related to #13

sameer, 2024-02-01:
Ahh I didn't see you already shared the SVG, I am using a dark theme for github :joy:

Will take a look at this soon

sameer, 2024-02-02:
I looked at your SVG more closely, this is a practical example of #13.