sameer/svg2gcode #50: Button disabled

tools-for-things-and-ideas, 2024-02-05:
Hello Sameer, 

I've been using the svg2gcode website a while. It's great!
But today the 'Generate G-code' stays greyed out (disabled). I tested this on multible browsers and computers (macs).
My apologies if I'm doing something wrong here.

All the best,
Elias Heuninck
http://www.formlab.schoolofarts.be/

sameer, 2024-02-05:
Weird, I think the latest release must've gone wrong. I'll look into this, thanks for letting me know!

sameer, 2024-02-05:
Haven't figured out what's wrong but I have reverted the relevant change, this should be fixed in a few minutes

closed 2024-02-05 by sameer

tools-for-things-and-ideas, 2024-02-06:
Thank you so much Sameer!

sameer, 2024-02-09:
The issue that caused this is fixed by 47e0301908494ca1f96f188e3cbd1bcfdbf25df8.