sameer/svg2gcode #51: Random Artefacts after Conversion of Custom Font

GordonLutz, 2024-03-01:
Hi Sameer, 

First of all thanks for this awesome Converter and thanks for making it public. I used your converter to convert svg into gcode for my PenPlotter. The issue im facing right now is that i purchased some new Single Line Fonts. Two of them seem to work fine after conversion. But I'm having trouble with one font. After converting to the Single Line Font i start the Converter and get random lines behind words. (see Screenshot). I already tried different words etc. but those artefacts seem to be random. Any Idea what that could be?

I also attached the Screenshot of another font to show that I dont encounter the problem with other fonts. I also asked the lead developer of AxiDraw as well as the support of Quantum Enterprise (wher i got the font from) and they told me that it is not a problem of the font but of the conversion. 

I hope you can help me there. 

Best regards
[image]
[image]
[image]

These are the files and resultes with the other font which dont yield those errors somehow

[image]
[image]
[image]

sameer, 2024-03-01:
Hi, this is actually the same problem as #49 & #13. 

Because the SVG segments are separate after conversion, they can be processed in an order that makes the gcode output do some seemingly random moves.

I'll see if I can come up with something to solve it this weekend since a few people have run into this now.

GordonLutz, 2024-03-01:
That would be awesome as help us out a lot!

sameer, 2024-03-04:
Hi @GordonLutz,

I misunderstood the issue while looking at the SVG on my phone. After looking at it more closely, I see the artifacts you are talking about in the NC Viewer output. The SVG you shared actually does contain these artefacts, though they do not draw anything and should not be visible.

```svg
<path d="M270 0 M540 270 " transform="translate(25.314349,-0.238672)scale(0.006285, -0.006285)" style="" id="az"/>
```

For instance, the above path consists of two moves. svg2gcode draws this path as those two move commands which NCViewer should render in orange, not blue. NCViewer is rendering these because there is a g-code comment between each path.

With the comment present, it renders the move in blue: 
[image]

After removing the comment and clicking plot, you can still see the artefact but it is orange instead: 
[image]

These artefacts should not affect the output on your pen plotter. But it does look like a bug in NCViewer that they are being rendered.

GordonLutz, 2024-03-06:
Hi @sameer, 
Thanks for looking into this issue and apologize my late response. 

I also found out that this temporarily fixes the issue but unfortunately doesnt fixes the problem while converting. I dont think its the best practice to manually remove each comment. Is there maybe another way to fix that? 

Best regards

sameer, 2024-03-07:
>  I dont think its the best practice to manually remove each comment. Is there maybe another way to fix that?

You could use another g-code viewer, I have used https://nraynaud.github.io/webgcode/ but the interface isn't great. I was thinking about creating a viewer myself but haven't gotten around to it yet 🙂

GordonLutz, 2024-03-08:
I found a workaround. If i define the Tool On/Off Sequence I can make it so the Tool is off when these artefacts happens. The maschine is running the path with G0 but not plotting the artefacts which is completely fine. But for my use case the usage of the CLI would be helpful. But i didnt found an documentation on how to use the CLI. Is there any table where I can see the syntax and the attributes i can define? Above all attributes like the programm start seqeunce, program end sequence as well as the Tool on and off GCode.

sameer, 2024-03-08:
> Is there any table where I can see the syntax and the attributes i can define? Above all attributes like the programm start seqeunce, program end sequence as well as the Tool on and off GCode.

Yes, if you run the CLI tool with `svg2gcode -h` it will show you how to use it:

```
svg2gcode 0.0.7
Sameer Puri <...>
Command line interface for svg2gcode

USAGE:
    svg2gcode [OPTIONS] [file]

FLAGS:
    -h, --help       Prints help information
    -V, --version    Prints version information

OPTIONS:
        --begin <begin-sequence>
            G-Code for initializing the machine at the beginning of the program

        --checksums <checksums>                              Include checksums at the end of each line
        --circular-interpolation <circular-interpolation>    Whether to use circular arcs when generating g-code
        --dimensions <dimensions>                            Override the width and height of the SVG (i.e. 210mm,297mm)
        --dpi <dpi>
            Dots per Inch (DPI) Used for scaling visual units (pixels, points, picas, etc.)

        --end <end-sequence>
            G-Code for stopping/idling the machine at the end of the program

        --export <export>
            Export current settings to a JSON file instead of converting

        --feedrate <feedrate>                                Machine feed rate (mm/min)
        --line-numbers <line-numbers>                        Include line numbers at the beginning of each line
        --origin <origin>                                    Coordinates for the bottom left corner of the machine
    -o, --out <out>
            Output file path (overwrites old files), else writes to stdout

        --settings <settings>
            Provide settings from a JSON file. Overrides command-line arguments

        --tolerance <tolerance>                              Curve interpolation tolerance (mm)
        --off <tool-off-sequence>                            G-Code for turning off the tool
        --on <tool-on-sequence>                              G-Code for turning on the tool

ARGS:
    <file>    A file path to an SVG, else reads from stdin
```

sameer, 2024-03-08:
Once you know the arguments you want to set, you can export them to a JSON settings file:

`svg2gcode --export settings.json'

```json
{
  "conversion": {
    "tolerance": 0.002,
    "feedrate": 300.0,
    "dpi": 96.0,
    "origin": [
      0.0,
      0.0
    ]
  },
  "machine": {
    "supported_functionality": {
      "circular_interpolation": false
    },
    "tool_on_sequence": null,
    "tool_off_sequence": null,
    "begin_sequence": null,
    "end_sequence": null
  },
  "postprocess": {
    "checksums": false,
    "line_numbers": false
  }
}
```

Which you can then use each time you call the program:

`svg2gcode --settings settings.json ....`

sameer, 2024-04-06:
Closing this as it's an issue with NCViewer, not svg2gcode

closed 2024-04-06 by sameer

sameer, 2024-04-08:
@GordonLutz I created an option just for this situation: 
[image]

So hopefully it's a little less postprocessing work for you :slightly_smiling_face: