sameer/svg2gcode #55: Conversion yields strange artefacts

GordonLutz, 2024-04-07:
Hi Sameer, 
i was casually using the converter yesterday but noticed very strange file conversions. Whenever i convert a file it starts adding places that dont even exist in my svg. I cant seem to wrap my head around the problem since a few days ago it worked without a problem. 

I attached a picture of how my files look as well as the corresponding SVG and a snippet of the generated G-Code
[image]
[image]
. 

Here is a snippet of how the converted GCode looks like: 
G21
G90;svg#svg1 > defs#defs1 > rect#rect50
G0 Z1 F5000
G0 X137.88582 Y-174.40669109375
G0 Z5 F5000
G1 X345.7205 Y-174.40669109375006 F7500
G1 X345.7205 Y-209.76203109374995 F7500
G1 X137.88582 Y-209.76203109374995 F7500
G1 X137.88582 Y-174.40669109375 F7500;svg#svg1 > defs#defs1 > rect#rect46
G0 Z1 F5000
G0 X313.95541 Y-105.81734109375004
G0 Z5 F5000
G1 X573.4635999999999 Y-105.81734109375009 F7500
G1 X573.4635999999999 Y-146.12242809375002 F7500
G1 X313.95541 Y-146.12242809375002 F7500
G1 X313.95541 Y-105.81734109375004 F7500;svg#svg1 > defs#defs1 > rect#rect45
G0 Z1 F5000
G0 X317.49094 Y-64.09804109375003
G0 Z5 F5000
G1 X574.8778100000001 Y-64.09804109375006 F7500
G1 X574.8778100000001 Y-101.57470009375002 F7500
G1 X317.49094 Y-101.57470009375002 F7500
G1 X317.49094 Y-64.09804109375003 F7500;svg#svg1 > defs#defs1 > rect#rect44
G0 Z1 F5000
G0 X323.14779999999996 Y-23.085841093750002
G0 Z5 F5000
G1 X602.45498 Y-23.085841093750002 F7500
G1 X602.45498 Y-61.26960709375 F7500
G1 X323.14779999999996 Y-61.26960709375003 F7500
G1 X323.14779999999996 Y-23.085841093750002 F7500;svg#svg1 > defs#defs1 > rect#rect43
G0 Z1 F5000
G0 X318.90516 Y26.41162890624997
G0 Z5 F5000
G1 X670.97578 Y26.411628906249938 F7500
G1 X670.97578 Y-10.443710093750022 F7500
G1 X318.90516 Y-10.443710093750022 F7500
G1 X318.90516 Y26.41162890624997 F7500;svg#svg1 > defs#defs1 > rect#rect41
G0 Z1 F5000
G0 X10.606602000000002 Y170.66141390624998
G0 Z5 F5000
G1 X292.999572 Y170.66141390624995 F7500
G1 X292.999572 Y-215.91989609374997 F7500
G1 X10.606602000000002 Y-215.91989609374997 F7500
G1 X10.606602000000002 Y170.66141390624998 F7500;svg#svg1 > defs#defs1 > rect#rect40
G0 Z1 F5000
G0 X16.263455999999998 Y152.27663790624996
G0 Z5 F5000
G1 X271.529006 Y152.27663790624996 F7500
G1 X271.529006 Y-188.54883209375 F7500
G1 X16.263455999999998 Y-188.54883209375 F7500
G1 X16.263455999999998 Y152.27663790624996 F7500;svg#svg1 > g#layer3 > g#g1 > g#a > g#au > path#ai
G0 Z1 F5000
G0 X3.5105521177896004 Y180.65737233945796
G0 Z5 F5000
G1 X3.5374946046796003 Y180.138091470428 F7500
G1 X3.5501734220396 Y179.716473517133 F7500
G1 X3.5517582742096003 Y179.37346230089298 F7500
G1 X3.5462112916146005 Y179.08841362813803 F7500
G1 X3.5367021785946 Y178.83988929785303 F7500
G1 X3.5264006394896 Y178.60803912391304 F7500
G1 X3.5192688047246006 Y178.37221891274808 F7500
G1 X3.5184763786396003 Y178.1117844707881 F7500
G1 X3.52620253296835 Y177.9434548924481 F7500
G1 X3.5462112916146005 Y177.7941814927881 F7500
G3 X3.5838515306521006 Y177.6651552829756 R0.6454224821019823 F7500
G3 X3.6444721261546005 Y177.55756727417813 R0.5026339845167382 F7500
G3 X3.73104467594085 Y177.47221147384062 R0.4071295715486516 F7500
G3 X3.8465407778296004 Y177.40988188940813 R0.5093163847072916 F7500
G1 X3.9164723798308505 Y177.38764968094816 F7500
G1 X3.9963093078946 Y177.37176953204818 F7500
G1 X4.086051562020851 Y177.36224144270818 F7500
G1 X4.1856991422096 Y177.35906541292817 F7500
G1 X4.313081635373349 Y177.36343245387567 F7500
G1 X4.454331585024599 Y177.37653357671817 F7500
G1 X4.611033843333351 Y177.39836878145564 F7500
G1 X4.784773262469601 Y177.42893806808814 F7500
G1 X4.9755498424333515 Y177.4688369421994 F7500
G1 X5.1833635832246 Y177.51866090937315 F7500
G1 X5.6564419559696 Y177.6464961080182 F7500;svg#svg1 > g#layer3 > g#g1 > g#a > g#au > path#ak

It starts going there weird places at the beginning and randomly throughout the file. I also tried that with different SVG Files that i converted a few days ago without a problem to see if my files have some bug in it but I get the same result as well and a different then a few days ago. 

Maybe you have an idea what happened. 

Best regards,

GordonLutz, 2024-04-07:
Here is also the picture of the SVG i attached. The Gcode correnctly adds my program begin sequence but then again makes really weird paths. 
[image]

sameer, 2024-04-08:
Hi Gordon,

Thanks for sharing your SVG. It contains some rectangles in the [defs](https://developer.mozilla.org/en-US/docs/Web/SVG/Element/defs) section:

```
    <rect x="137.88582" y="357.08892" width="207.83468" height="35.35534" id="rect50"/>
    <rect x="313.95541" y="288.49957" width="259.50819" height="40.305087" id="rect46"/>
    <rect x="317.49094" y="246.78027" width="257.38687" height="37.476659" id="rect45"/>
    <rect x="323.1478" y="205.76807" width="279.30718" height="38.183766" id="rect44"/>
    <rect x="318.90516" y="156.2706" width="352.07062" height="36.855339" id="rect43"/>
    <rect x="10.606602" y="12.020815" width="282.39297" height="386.58131" id="rect41"/>
    <rect x="16.263456" y="30.405591" width="255.26555" height="340.82547" id="rect40"/>
```

With the latest release, svg2gcode now renders shapes. Because I never implemented handling for `defs`  those shapes are just being rendered (incorrectly).

closed 2024-04-08 by sameer

sameer, 2024-04-08:
Generated gcode for the image you shared and it looks good to me now:
[image]

commit 6a1e23f6 (2024-04-08): lib v0.2.2: fix #55