sameer/svg2gcode #57: How to ignore markers

MadTooler, 2024-07-18:
Can I option through the CLI to ignore start and end markers of paths?

Thanks

sameer, 2024-07-18:
Hey, I don't think there's a setting for this yet. The comments were mostly intended as a debugging aid so I think it makes sense to have one.

MadTooler, 2024-07-18:
Thanks for the reply.

Not sure if we are thinking of the same thing.  Are you referring to the arguments that can be passed in the CLI as the "comments?"

Take a look at the attached svg with triangle marker-start and square marker-end.  Give it a try with your svg2gcode and you will see it adds the markers separately after the path they were to be applied, and removed from them.  In my case, the markers are being used as indicators for machine direction by my custom gcode sender.  I don't actually want them to be processed by svg3gcode.
[image]

sameer, 2024-07-18:
Sorry for the confusion, I thought by markers you were talking about the GCode comments in the output.

If you want to ignore those shapes, you can set `style="display:none` on them and svg2gcode will skip them https://github.com/sameer/svg2gcode/blob/a2daff9d9716c6e5f44b07e3e176a0cc7b43b366/lib/src/converter/visit.rs#L39

MadTooler, 2024-07-19:
No problem.  I appreciate your quick responses.

I have tried to apply the  --style="display:none"  in the CLI.  Apparently I am doing it wrong.  What is the proper format?

MadTooler, 2024-07-19:
Oh, I think I understand the style to be an additional parameter in the svg instead of the CLI argument I was assuming.  I will try that and see if it works.

MadTooler, 2024-07-19:
Well, adding the style display none in my svg to the markers does keep them from being output by svg2gcode, but they are also no longer displayed in my svg viewer.  I can do a work around if that is the limitation, just trying to see what options there are before I go that route.
Thanks again.

sameer, 2024-07-19:
Yeah svg2gcode will try to draw everything it is given in an SVG. Maybe we can come up with some special SVG element marker to indicate that they should be skipped by svg2gcode but will still display otherwise?

MadTooler, 2024-07-19:
I don't know how it would work for others, but my first thought is a command line argument maybe similar to "--ignore foo" where foo is an id of a path.  Or maybe and option to ignore any paths with either a prefix or suffix the ignore looks for?  This seems like it could be a bit unreliable.  Just my first thoughts.

sameer, 2024-07-19:
> I don't know how it would work for others, but my first thought is a command line argument maybe similar to "--ignore foo" where foo is an id of a path.  Or maybe and option to ignore any paths with [...]

I think it's a good idea. We could do it as a [CSS selector](https://developer.mozilla.org/en-US/docs/Web/CSS/CSS_selectors/Selectors_and_combinators) so it's easy to write. I'll start looking into this

MadTooler, 2024-08-06:
I found svg's created in inkscape with stroke end markers do not result in svg2gcode outputting the end markers.  I also created some reference geometry via an inkscape plugin "inkstitch" that also did not get output to gcode.  

My heart file I posted earlier was originally made in inkscape without end markers.  I later added them through some XML editing via an svg node in NodeRed.  Something about the format I used causes the markers and other reference paths to be seen and output by svg2gcode where it is ignored when direct from inkscape.

sameer, 2024-08-06:
> I found svg's created in inkscape with stroke end markers do not result in svg2gcode outputting the end markers. I also created some reference geometry via an inkscape plugin "inkstitch" that also did not get output to gcode.

Interesting, could you share those SVGs? I wonder if the marker is in a `defs` section. In which case the fact that svg2gcode doesn't support use/defs (#56) is probably why.

> My heart file I posted earlier was originally made in inkscape without end markers. I later added them through some XML editing via an svg node in NodeRed. Something about the format I used causes the markers and other reference paths to be seen and output by svg2gcode where it is ignored when direct from inkscape.
> 
> While I need to figure out what I am doing that may be undesirable, I think some sort of direct method of ignoring certain paths from the CLI would be desirable.

The marker support is looking pretty broken, reading the [MDN page](https://developer.mozilla.org/en-US/docs/Web/SVG/Element/marker) there's a lot of missing things needed to render them appropriately. For now, I can disable markers entirely until they are supported and file a separate issue for that.

In the meanwhile I'll find some time to also try and figure out ignoring/skipping. It's looking more involved than I thought..

MadTooler, 2024-08-06:
> Interesting, could you share those SVGs? I wonder if the marker is in a defs section. In which case the fact that svg2gcode doesn't support use/defs (https://github.com/sameer/svg2gcode/issues/56) is probably why.

Yes, the stroke style end marker paths are in defs.  However, the origin generated by inkstitch is in layer1.  

See this version from inkscape.  For some reason only one quadrant of the inkstitch origin vector is shown here on github or with your web UI. 

[image]

sameer, 2024-08-06:
> > Interesting, could you share those SVGs? I wonder if the marker is in a defs section. In which case the fact that svg2gcode doesn't support use/defs (#56) is probably why.
> 
> Yes, the stroke style end marker paths are in defs. However, the origin generated by inkstitch is in layer1.

That's definitely why then. I'll add in a change to skip markers for now until they are properly handled.

> See this version from inkscape. For some reason only one quadrant of the inkstitch origin vector is shown here on github or with your web UI.

Weird, the GitHub preview looks completely different for me compared to on Firefox/Chrome. How it looks in the web UI is just whatever the browser you're using would do if you opened it. 

[image]

MadTooler, 2024-08-07:
> That's definitely why then. I'll add in a change to skip markers for now until they are properly handled.

Skipping markers is fine by me.  I am sure someone may want to have them somehow magically converted to gcode, but that is a loaded task.  For my use, I like them being ignored.

Digging into this, I realized my code generated svg's should have also had the markers in the defs, but they were not making it in.  If not for my bad format, I would never have known to bring this issue up.

Moving forward, I am injecting "home" or "origin" single line segment paths with arrow marker start or as a symbol via "use id" to create an origin reference that is not at viewbox bottom left.  From there, when I import the svg into my gui, I am trying to parse the xy location of the "home" and calculate the origin offset to feed svg2gcode from there.  After gcode generation, if I end up with a simple path segment "home" reference, I parse and remove the related lines after the "home" path id comment is seen in the gcode.  That's the long way around getting everything I need, but it should mostly work OK.

Thanks for all your time and work.

sameer, 2024-08-08:
I looked at the SVG spec a little more closely today. Neither markers nor symbols are meant to be rendered directly, so I've gone ahead and stopped svg2gcode from rendering them.

Hope this helps!

MadTooler, 2024-08-08:
So it looks like for now I am running smoother in regards to the marker issue, as long as my formatting works as intended.

I do still think an option to mark items for ignore would be very useful.  Not sure how you are parsing the svg, but from a user perspective, ideally simple as adding a prefix/suffix to the path or object name to ignore would be slick.  More involved, again from user, would be to add some text label inside the object in the xml.  From within inkscape, that could be either through object properties, layers, or xml editor.  Layers or Properties would be the easier from user side.  Not sure the most universal for other vector software.

In any case, glad you were able to get things moved further along.  My system is definitely benefiting from your work.

Thanks!!