sameer/svg2gcode #58: M2 End of program always appended

tools-for-things-and-ideas, 2024-07-29:
As discussed here ( https://github.com/sameer/svg2gcode/issues/23#issuecomment-953073378 ) the appended 'M2' (End of program) is correct. The way the servo on my pen plotter reacts to the M2 is correct too: it moves the servo to the zero-position. In my particular case that's not the best result as it leaves the pen on the paper.
I can work around this by making a Gcode post processor or by making the plotter move outside of the paper area in the 'Program End Sequence'. Another solution would be to remake the pen plotter holder so the servo moves the pen up when it's at zero. 

Alternatively the SVG2Gcode website could put the M2 in the suggested contents of the Program End Sequence, so people can deactivate it when they would like to. This makes the G-code generation also a bit more transparent, as the resulting G-code would be more predictable. To be clear: this is not a criticism or a bug report. Just a suggestion. I fully understand if there are reasons to keep the website working the way it does now. It's a great tool!

Elias

petervanderwalt, 2024-07-29:
[image]

@sameer could change this setting to overwrite his chosen defaults (enter the M2 etc in the field by default, but allow users to edit it not wanted) 

I tried adding my own expecting it to overwrite

[image]

But sadly that just gets prepended to the M2

[image]

Not as configurable as the description of the form field "G-Code for stopping/idling the machine at the end of the program" implies if it adds extra gcode after

sameer, 2024-07-29:
> Alternatively the SVG2Gcode website could put the M2 in the suggested contents of the Program End Sequence, so people can deactivate it when they would like to.

Hey Elias, I think this is a great idea! Thank you for sharing the context around it.

To make this change go smoothly, I'm planning to upgrade the config file in a way that would append M2 to the program end sequence the next time you load the web version of svg2gcode.

On the CLI side, keeping things consistent is a little more tricky. An end sequence in a config file can still have M2 inserted until the format is upgraded. For an end sequence passed as an argument, it won't be obvious that M2 is no longer included. I'll call it out in the release notes.

tools-for-things-and-ideas, 2024-07-30:
Hi Sameer, 

Thank you for your swift reply!
I see it's not a straightforward change. No hurry for the update!
Sounds exciting though! Good luck!

sameer, 2024-08-06:
This should be fixed by [0c5717f](https://github.com/sameer/svg2gcode/commit/0c5717f0c013dec6a26b66b343bdfa585da9c885), please refresh the web version and give it a try :slightly_smiling_face:

tools-for-things-and-ideas, 2024-08-07:
I did a first test which went great!
Thanks for the quick update!

closed 2024-08-07 by sameer