sameer/svg2gcode #65: resulting gcode positions way too small

tcurdt, 2024-12-08:
The resulting gcode for this svg has dimensions that are 0.00something.
I assume svg2gcode does not like the width/height/viewbox setting here?

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg width="100%" height="100%" viewBox="0 0 2481 3508" version="1.1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" xml:space="preserve" xmlns:serif="http://www.serif.com/" style="fill-rule:evenodd;clip-rule:evenodd;stroke-linecap:round;stroke-linejoin:round;stroke-miterlimit:1.5;">
    <rect x="414.339" y="605.03" width="1722.03" height="1487.66" style="fill:none;stroke:black;stroke-width:1px;"/>
    <path d="M787.051,991.288C787.051,991.288 1281.03,963.267 1402.26,1022.86C1505.39,1073.56 1514.44,1348.86 1514.44,1348.86L919.849,1254.57" style="fill:none;stroke:black;stroke-width:1px;"/>
</svg>
```

```
G21
G90;svg > rect
G0 X0.022393681679969593 Y0.07006408019764349
G1 X0.06395540916001521 Y0.07006408019764349 F300
G1 X0.06395540916001521 Y0.03415894526795895 F300
G1 X0.022393681679969593 Y0.03415894526795895 F300
G1 X0.022393681679969593 Y0.07006408019764349 F300;svg > path
G0 X0.03138920125427594 Y0.06074162371721779
G1 X0.03771610979665527 Y0.06089998453534777 F300
G1 X0.04623746104142911 Y0.059979623717217774 F300
G1 X0.04894496009122006 Y0.052111512732801214 F300
G1 X0.03459432212086659 Y0.054387231090839976 F300
```

sameer, 2024-12-08:
The viewBox looks fine to me. I'll debug this tomorrow, but in the meantime a few things you can try removing the 100%/100% width/height attributes. Set those to physical units if needed (in/cm).

tcurdt, 2024-12-09:
This seems to work just fine.

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg width="210mm" height="297mm" viewBox="0 0 2481 3508" version="1.1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" xml:space="preserve" xmlns:serif="http://www.serif.com/" style="fill-rule:evenodd;clip-rule:evenodd;stroke-linecap:round;stroke-linejoin:round;stroke-miterlimit:1.5;">
    <rect x="414.339" y="605.03" width="1722.03" height="1487.66" style="fill:none;stroke:black;stroke-width:1px;"/>
    <path d="M787.051,991.288C787.051,991.288 1281.03,963.267 1402.26,1022.86C1505.39,1073.56 1514.44,1348.86 1514.44,1348.86L919.849,1254.57" style="fill:none;stroke:black;stroke-width:1px;"/>
</svg>
```

I blame the tool for outputting the 100%.

tcurdt, 2024-12-09:
Maybe it would be good to complain and show an error.

tcurdt, 2024-12-16:
Hm - but now with...

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!-- Created with Inkscape (http://www.inkscape.org/) -->

<svg
   width="70mm"
   height="70mm"
   viewBox="0 0 70 70"
   version="1.1"
   id="svg1"
   inkscape:version="1.4 (e7c3feb1, 2024-10-09)"
   sodipodi:docname="foo.svg"
   xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
   xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:svg="http://www.w3.org/2000/svg">
```

I would expect it come out the correct size.

What dpi does this need when the exact dimensions are given?

tcurdt, 2024-12-16:
I actually had to interpolate the dpi to 95.63 for it to come out correctly in the correct size 🤔

sameer, 2024-12-16:
> I actually had to interpolate the dpi to 95.63 for it to come out correctly in the correct size 🤔

Can you share the image?

tcurdt, 2024-12-17:
Not sure whether it's OK to attach here - but I could sent you a copy over a more private channel if that's OK with you?

sameer, 2024-12-17:
> Not sure whether it's OK to attach here - but I could sent you a copy over a more private channel if that's OK with you?

Sure, you can email it to me at the email here: https://purisa.me/about/