sameer/svg2gcode #73: my gcode has a Y axis offset

MaierJuerg, 2025-03-08:
In ubuntu/imageViewer my SVG file shows the image size as 212.4pt x 984.24pt.
This should result in X and Y values in the range 75mm x 347mm
When I create the gcode it adds 87 mm (rounded) to all Y positions, therefore running over the edge of my sheet
Any settings to remove this offset or explanation why it adds this value?

sameer, 2025-03-08:
Hi, can you share the SVG you're using?

MaierJuerg, 2025-03-09:
Hi Sameer
I apologize for my mistake. I have been using a program to split my svg
into parts and I found out now that this modified my svg. Rerunning the
original svg shows no Y-offset.
Many thanks for your quick reply
Regards
Juerg Maier

sameer, 2025-03-09:
No problem, glad we figured this out!

closed 2025-03-09 by sameer